lolwheel/Owie #4: D1 Mini lite version Compatibility

TheBR, 2022-01-17:
Hi,
  Im from the UK and have been keeping an eye on this as i have a ChiBattery modded 4210/4150 with limited range due to the firmware version.

Its, difficult to source a WeMos D1 Mini Lite here. Well, actually what i mean by that is that its difficult to source a V1 as you have in the wiki.

It is possible to source D1 mini V3.0.0's and to my eyes they look identical, apart from having 4mb for example - https://www.ebay.co.uk/itm/313603433691?hash=item490438a4db:g:cFQAAOSwwMBf7MQa

Can you advise if this is compatible? The size and shape look the same so i expect it would fit but i dont know much about these chips so i thought i would ask.

Finally, is this ready do you think yet?

Many Thanks for your work on this project.

lolwheel, 2022-01-17:
Yup It should work on any version of D1 as long as it's ESP8266 based.

The project isn't of a particular usefulness as of now, I have defeated the BMS pairing so far which isn't something you require. Keep an eye on it, I'll be getting started on battery stuff soon.

Cheers

-------- Original Message --------

TheBR, 2022-01-17:
No problem,
  Many thanks for the info and the very best of luck with the battery stuff! Fingers crossed for you.

closed 2022-01-20 by lolwheel