lolwheel/Owie #19: Is there any tutorial on how to make the firmware run on the board - if one is not ESP8266 familiar

haemse, 2022-02-22:
which software suite
what files to chose ...
or a link to a tut where you see how to do it.

think a few descriptive keywords in the readme would already do it.

i would write it if i knew how to do it

would be great

ow-breaker, 2022-02-23:
You can flash the firmware on this site: https://ow-breaker.github.io/

haemse, 2022-02-24:
Directly from the browser to the board via USB Cable? How cool is that?

rgalverson, 2022-03-01:
@ow-breaker thanks, that worked.  However, if I want to roll my own version of firmware, do I need to use the ESPressif download tool?  I'm getting a "5-chip efuse check fail" from that (ESP8266 V3.9.2).  "TheChickenTendy" posted a video on building and downloading that almost worked for me.

ow-breaker, 2022-03-01:
> 

I've had success using [this tool](https://github.com/marcelstoer/nodemcu-pyflasher), but if you're thinking about building and deploying your own firmware, maybe you should get comfortable with googling around first. 

This project is for two types of people, those who want a cheap alternative to JWFFM and are happy with the stock features (and will flash with the web gui) or those who want JWFFM and want to be able to tinker with the thing themselves. For both cases there is **no** support, but especially in the latter case it's assumed that you have some knowledge of hardware and embedded development, as I (and I'm sure @lolwheel) are not too interested in helping every single person out.

The docs are quite good and do a fair job of explaining things, and the code quality is quite high, so it's fairly easy to read; at least for someone experienced.

rgalverson, 2022-03-01:
Thanks for the quick response.  I've used the Arduino environment before
with esp32, so I know enough to be dangerous.  The owie code is
straightforward, and not too different from the ino sketches I have used
before.

On my list for enhancements is support for two wifi networks, so I can have
both my home and my phone tether in there without having to switch.  I did
already Google and find someone doing that with esp32...

I'll give the tool you mentioned a try.

Thanks

closed 2022-03-19 by lolwheel