lolwheel/Owie #23: Using a Onewheel Pint BMS in an XR with the OWIE chip

wxyzsupermod, 2022-02-28:
I notice in the code there is a spoof function for the serial number is there any reason we cannot use a onewheel pint BMS in an XR the XR bms is broken and I think that the pint and the XR work off the same info so as long as I spoof the serial it should work correct?

lolwheel, 2022-02-28:
Correct, Pints and XRs use the same BMSes, serial spoofing is in place exactly for that.

Sent from ProtonMail mobile

-------- Original Message --------

wxyzsupermod, 2022-02-28:
So I will need to recompile the firmware to use that correct? set serial_override_ = XR SERIAL; defined in bms_relay.h correct?

lolwheel, 2022-02-28:
Nope, the option is available through the WiFi based configuration UI

Sent from ProtonMail mobile

-------- Original Message --------

wxyzsupermod, 2022-02-28:
Awesome will close this once attempted

dizzydonut10, 2022-03-04:
After installing the chip and powering on before screwing things back together im getting an error 16. 

Im fairly certain there are no shorts between pins on the chip, and I don't believe i have any pinched or damaged harness cables. 

Im wondering if im misunderstanding the process for spoofing the serial and that's my issue. 

The serial on the broken xr bms is: 21100221325 
Inputting this number into the owie app causes it to display a completely different serial after reboot, and after looking at comments in the code Its my understanding that I need to remove the first 4 significant digits before i enter it, making the serial: 0221325

When i enter that and reboot, I noticed it doesn't completely change the serial this time, but instead drops the first zero making it: 221324

Any clarification on the spoofing process would be really helpful, thank you!

wxyzsupermod, 2022-03-07:
> Nope, the option is available through the WiFi based configuration UI Sent from ProtonMail mobile
> […](#)
> -------- Original Message --------

I believe this is the issue is in network.cpp line 192         Settings->bms_serial = bmsSerialParam->value().toInt();
I believe the 0 is dropped when the string is converted to an int but I do not have a onewheel currently to test

lolwheel, 2022-03-19:
Github issues is no place for personal troubleshooting guys. Closing this issue. Pop over to the modding thread on OW Discord, you'll get plenty of support there.

closed 2022-03-19 by lolwheel

nataliegroman, 2022-08-10:
I'm sorry to open this back up, but at the moment, I am having the same issue. While XR and Pint use the same BMS, there's some sort of other signal that continues to give me an error 16 despite having installed it correctly (and I also reflashed a second chip to double check). Everything is reading perfectly, it's just that there's apparently, per JWXR, another heartbeat signal that Pints and XRs do not share; spoofing a serial in this instance is not enough. Some sort of way to figure out this interfacing would be fantastic.