lolwheel/Owie #25: Buzzer support

bnmoore, 2022-03-11:
Nice work on this project! I am planning to modify your code to enable a buzzer which would warn of impending pushback, i.e. when battery-to-board current goes above a certain threshold. Are you thinking to add this type of feature at some point?

ow-breaker, 2022-03-12:
So this is an idea that I've been kicking around. I'm not certain how FM is doing it on the GT, but I'd suspect that they're actually using the signaling to the motor driver circuit to tell when you're approaching the limits of the board, opposed to battery current.
Using the signals is far more accurate and reliable as that tells you the instantaneous load on the controller and gives you how close the system is to reaching it's limiting factor.
However, as a primitive system, using battery current could give you a 'good enough' idea of how close your board is to the limit. It won't be perfect, but it'll likely give users a decent idea of when they're getting close (say 75-80%).

bnmoore, 2022-03-12:
I just finished installing the chip, it is a tight fit! There was not enough room to mount the buzzer directly to the board so I ran it out towards the charger port (buzzer is in red shrink-wrap at the bottom of the picture). I opted to attach the buzzer to 3.3V and D7 (GPIO13). I put a startup beep sequence in the code and so far it seems to be working. The buzzer I had on hand is a bit too quiet unfortunately. I don't know if a louder buzzer would still be able to fit in the case. Regardless, I plan to do some testing next week. I am on the Pint so I set the code to beep when the current goes above 10A. We'll see how it goes.

[image]

lolwheel, 2022-03-19:
Sorry I got no plans to support this, I am not convinced that Owie/BMS information is enough to provide what you're trying to achieve.

10 amps at full charge with backing of a powerful battery or a VnR setup is different from 10 amps at 45 volts. What you really need is duty cycle of the PWM powering the motor.

Another issue you've bumped into yourself is the power needed to support any sort of loud buzzer. The BMS power supply doesn't have much juice. This was the reason behind me using ESP8266 as opposed to smaller ESP32 boards - The BMS would die with the WiFI modem turning on because it didn't have enough juice

closed 2022-03-19 by lolwheel

bnmoore, 2022-03-20:
Yep, I am abandoning this as well. Would be more flexible to build this feature into an app if ever. Pushback is always there as a warning and I don't think the owie chip can prevent pushback. 

To clarify regarding beeper volume, the beeper itself is plenty loud and only draws <25 mA according to the spec sheet. The issue is that the sound is muffled by the waterproof casing of the Onewheel battery compartment. If there was a good way to mount outside the case then this would be less of a problem.