lolwheel/Owie #26: OWIE  - Overridden SOC never under 5%

wheelerone, 2022-03-19:
Hi, Onewheel Pint (5314 - Gemini 5050) with Quart Battery will never Captain Morgan (low battery). 
Lowest Overridden SOC is 5%, what should i do?
Regards

[image]

lolwheel, 2022-03-19:
Yeah just ate it myself because of this, fixing.

lolwheel, 2022-03-19:
Fixed.

dmimwheel, 2022-03-20:
perfect, 5 % floor fixed! Light bar still yellow, not going red at low volatge and no captain morgan :/

lolwheel, 2022-03-21:
OK thanks for the feedback. I think I have an idea what's going on, it's just that I'm not sure I'll be able to fix it and this isn't too high up on my priority list.

I'm fairly confident that the "battery empty" flag is part of the status message (message code 0) which I am swallowing completely to circumvent the range limitation on the newer boards. I need to figure out which flag in the status byte does indeed correspond to this state and let a packet through when I see this flag set.

For this I'll have to run the board battery into the ground and I'm sorta lazy to do that.

If you guys end up in that situation, you can help me out by opening the Owie web config page, hit "Monitor BMS Data" button and save the first line in the text box which will look something like "ff 55 aa 00 XX 02 YY", the XX or the 5-th byte is the only thing that matters.

Imma reopen this bug, looks like we do have some work to do.

wheelerone, 2022-03-22:
here is the Monitor BMS Data :)
[image]

lolwheel, 2022-03-27:
Yay teamwork! Never seen the 0x4 flag appear before, just tried injecting it and tadaa, got the pushback and "Needs juice" notification in the app, that was it!

Submitting the fix in a sec.

closed 2022-03-27 by lolwheel