lolwheel/Owie #27: Wemos D1 mini is the smallest really?

Warialinth, 2022-04-02:
Hi, reading trough the readme and comments people have on this project I started to wonder if ESP-01 module would be much better fit. It is much smaller board that would be better fit into already crammed compartment.
Just an idea.

Warialinth, 2022-04-02:
Correction to not sound rude: Can you add support (pinout change) for ESP-01 boards as they have different pins avaliable?
[image]

lolwheel, 2022-04-02:
Won't work, it lacks the power regulator.

Sent from ProtonMail mobile

-------- Original Message --------

Warialinth, 2022-04-02:
I would dare to say that there has to be 3V3 somewhere on the BMS board or if not then AMS1117 module is also way to go without too much space needed.
(Can't probe around my own board because my pack has multiple shorted cells from water damage.

lolwheel, 2022-04-03:
Yes there likely is a 3.3v regulator on BMS but no one will be soldering wires on SMD contacts.

The rest of your request doesn't make much sense to me either.
Ams1117 externally on an esp-01 to save space? Esp-01 will require an external programmer as well to flash it the first time, this won't be a user friendly setup.

Sounds like you're somewhat comfortable around electronics so if you insist on this, adapting the project to esp-01 will require trivial if any changes to the source code.-------- Original Message --------

Warialinth, 2022-04-03:
That is the problem being good at soldering on professional level doesn't mean one can do much on software side of things, wasn't been able to even find where or what values to change to modify the code myself...

The combo of the AMS1117 and ESP-01 boards are still smaller than JW chip.

closed 2022-04-03 by Warialinth