lolwheel/Owie #29: BMS reported SOC always at 0%

FAB1150, 2022-04-09:
As the title says, the "BMS reported SOC" is always at 0% in the Owie web interface. I have plugged the board in, rebooted it, and ridden it for a bit, but it stays at 0%.

I have verified that my wiring is correct, and the Owie does report the charge to the main board (aka "overridden SOC" is correctly displayed  both in the web interface and the OW app)

As I cannot read what's in the "monitor BMS data" page to tell if it's a problem with the BMS or with the Owie, I'll attach it under here.

Main page in the interface:
[image]

Monitor BMS data page:
[image]

FAB1150, 2022-04-09:
I forgot to mention that the battery is the stock XR battery, in an XR.

mretich, 2022-04-10:
Hi, have you charged your Board in the meantime?
Because my guess is, that it has probably something to do that BMS goes through a state reset and doesn't know the status of the battery, and plugging the board into a charger corrects this issue by forcing the BMS (and controller potentially) to do a state check.
(Like the problem with the 1% Battery reporting)

FAB1150, 2022-04-10:
That was my first thought too at first, but I have put it into charge a few times, and it still reports 0%. I have also done a few miles on it now. 
I have not charged it to a full 100% or discharged it to actual 0% yet though, should I try to do it?

lolwheel, 2022-04-10:
Try charging the board fully. Your board would have shown 0% even if you completely removed Owie. This is the BMS itself acting up.

One of the users has seen a behavior like this. He said it was fixed by fully charging. See https://github.com/lolwheel/Owie#board-reporting-battery-at-1-after-install

closed 2022-04-10 by lolwheel

FAB1150, 2022-04-13:
Sorry for the delay, but charging to 100% did fix it. I don't know why charging it to 70-80% didn't do anything before, but it now works flawlessly. Thanks!