lolwheel/Owie #31: LED support?

matzZz, 2022-04-21:
Hi,
is it possible that you add support for battery level shown on a Adafruit NeoPixel Stick (Like Pint/GT status light)?

lolwheel, 2022-04-21:
It'd be pretty hard as ws2812b lights require tight timing for updates and esp8266 doesn't have a dedicated pwm so I wanna say no.

-------- Original Message --------

matzZz, 2022-04-21:
A led like this should work…
http://www.esp8266learning.com/some-neopixel-examples.php

ow-breaker, 2022-04-21:
It's probably possible using the FASTLED library, however you have to be
extremely cognizant of how long it'll take to write the data to the LEDs as
that can be a point of potential issues.
I'm unsure of what board you're planning to do this on, but you'll have to
run some wires out of the battery box, which requires pretty serious
modification and forethought.

lolwheel, 2022-04-25:
This isn't viable, closing the issue.

closed 2022-04-25 by lolwheel