lolwheel/Owie #32: Realtime updates in browser window?

vesters7, 2022-04-24:
Is it possible to have realtime updates when connected to the browser window on a phone? 

Instead of having to refresh the page to check balancing and all that?

mretich, 2022-04-25:
Hi @vesters7,
i have done this on my fork already.
I have now created a pull request, so maybe this is going to be implemented here.
(Along with some other features, i think are usefull)
We will see ;)

vesters7, 2022-04-25:
@mretich i may try and flash your fork then for now, ( my cbxr has some weirs cells and im trying to moniter them under different loads)

mretich, 2022-04-25:
@vesters7: you should probably flash the new firmware either via usb cable or in the OTA recovery mode.
My last flash without recovery OTA left the chip acting weird... (e.g.: had some hard time showing the webinterface..)
After a "recovery" firmware update (e.g.: setting the chip in recovery mode => 3x power off and on the onewheel) everything was normal again.
Just to mention it ;)

mretich, 2022-04-27:
@vesters7: I recommend to use this repository again, since my changes are now merge in here with some important code improvements!
@lolwheel: I think this can be closed now :)

vesters7, 2022-04-27:
> now

So just reflash with the new build, i dont see any releses for firmware on the original

lolwheel, 2022-04-27:
I haven't set up official releases. You'll have to build it yourself. Open the project in GitPod and download the firmware, I'm pretty sure this process is described in the landing page README

closed 2022-04-27 by lolwheel

mretich, 2022-10-12:
> Totally possible, frontend work isn't my favorite though so this isn't high on my priority list
> […](#)
> -------- Original Message --------

@lolwheel : 
I'm a little confused about your comment on this old closed issue :)
This is already done in the merged PR #33..

Btw. a friend and I are currently working on a complete frontend design overhaul :)
(https://github.com/mretich/owie-design-update-mockup)