lolwheel/Owie #37: Did not charge

matzZz, 2022-05-10:
since i installed owie…
if i take a x minute break while a ride the app thinks i'm charging the Onewheel.
[image]

AnfroneyD, 2022-05-11:
When you stop riding, the voltage slowly creeps up and stabilises, it will appear as charging on the app. There's no issue here.

Eg: when riding your voltage would go down to 57v, once you stop putting load on the battery by stopping they will slowly rise back up in voltage. Hence showing charging

AnfroneyD, 2022-05-11:
It's just the way the owie works, it shows percentage left off voltage, the bms appears to show battery remaining as Amp Hours used,

That's why it fluctuates up and down as you ride

lolwheel, 2022-05-11:
@AnfroneyD is correct. This is just an artifact of Owie using only voltage to guesstimate the charge state.

closed 2022-05-11 by lolwheel