lolwheel/Owie #42: After platformio.ini changes the OTA does not work anymore.

mretich, 2022-05-30:
Hi,
after your changes in the platformio.ini file, the OTA Update returns with an 400 HTTP Error containing following response: "OTA could not begin".

I was tracking this issue down to the change where you switched a lib_dep from "https://github.com/me-no-dev/ESPAsyncWebServer@^1.2.3"  to "me-no-dev/ESP Async WebServer@1.2.3"

After the change the lib loaded "ESPAsyncTCP@1.2.2" AND "AsyncTCP@1.1.1"
while the old dependency only fetches "AsyncTCP@1.1.1" and omitting the "ESPAsyncTCP" library...

Here is a trace using "me-no-dev/ESP Async WebServer@1.2.3"
_Library Manager: ESP Async WebServer@1.2.3 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing ESPAsyncTCP
Unpacking  [####################################]  100%
Library Manager: ESPAsyncTCP@1.2.2 has been installed!
Library Manager: Installing AsyncTCP
Library Manager: Warning! More than one package has been found by AsyncTCP requirements:
Library Manager:  - me-no-dev/AsyncTCP@1.1.1
Library Manager:  - kubafilinger/AsyncTCP@1.1.1
Library Manager: Please specify detailed REQUIREMENTS using package owner and version (shown above) to avoid name conflicts
Unpacking  [####################################]  100%
Library Manager: AsyncTCP@1.1.1 has been installed!_

and here the trace using "https://github.com/me-no-dev/ESPAsyncWebServer@^1.2.3"
_Library Manager: ESP Async WebServer@1.2.3+sha.f71e3d4 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing me-no-dev/AsyncTCP @ ^1.1.1
Unpacking  [####################################]  100%
Library Manager: AsyncTCP@1.1.1 has been installed!_

I know the ESPAsyncTCP IS used in the code, and NOT the AsyncTCP (since the later one is not used in esp8266
 afaik) and I really have NO clue why it still works even when the ESPAsyncTCP is NOT loaded as dependency...
 I guess it is included in the GIT (dev) version, but NOT in the Release ?

Nevertheless, this change causes a file size change of the resulting firmware binary from ~447k to ~645k.

I have no clue if I´m correct here, but I think this is the issue why ElegantOTA is not able to flash the binary file...

lolwheel, 2022-05-31:
Hey appreciate you doing all the digging. I'm fighting the same issue myself right now though I found some other ways to reduce the binary bloat. I'm on it, cheers!

lolwheel, 2022-05-31:
Fixed.

closed 2022-05-31 by lolwheel