lolwheel/Owie #43: Overridden SOC Not matching BMS Reported SOC after full overnight charge.

an-d-uu, 2022-05-31:
I am trying to see if this is an issue or by design.  The BMS reported SOC and the Overridden SOC are not the same value even after leaving the onewheel pint on its standard charger overnight. I updated to version 1.1.0 this morning to see if that improved the reported value but am not seeing the values match when I believe they should. I think the piece that I am missing is when you are not using a third party battery or external battery like the GT40 are the BMS reported SOC and Overridden SOC supposed to be different?

[image]

I would like to contribute to the project, I am just having a bit of difficulty getting up to speed. My setup is original BMS, original main board, and stock battery. I purchased the GT40 mod which is one of the main reasons I wanted to use the OWIE chip to unlock battery options, but am not seeing the GT40 battery percent being registered on the lights on the board. I ran the board's battery all the way down to 0 last night with the GT40 still attached and could continue riding with the OWIE app reporting I still have 56% battery left, but the board its self reports 0 with the yellow low battery light staying on.

danilo-dreia, 2022-05-31:
It looks like there is an issue with your battery. There are two cells that deviate from the rest making your BMS believe that it's charged more than it is. Owie is working correctly and is reading the battery voltage and converting it to a calculated SOC. If it passed the BMS reported SOC it wouldn't work correctly with range extenders.

closed 2022-05-31 by lolwheel

an-d-uu, 2022-06-01:
Thank you all for responding, this have been really helpful! I am now thinking it may also be that charger is not working correctly. I noticed that after a 10 to 15 minutes of charging it turns off and stops charging, which would actually explain why the cells aren't balancing properly.