lolwheel/Owie #45: BMS reports 0%, Overridden SOC reports 67%, and the Battery has 56.88v, but the board won't let me ride

an-d-uu, 2022-06-04:
I am trying to diagnose this issue and will include a screenshot. Basically the BMS reports 0% which after a few seconds shuts the board down even though the battery voltage is at 56.88v. Where should I be looking to find the issue?

[image]

an-d-uu, 2022-06-04:
After I posted this I saw issue #29 and am in the process of charging this to 100%, will report back if it does it again once it's finished charging.

lolwheel, 2022-06-04:
what hw revision of the board? the reported soc should have nothing to do with shutting off. likely that's your BMS acting up

-------- Original Message --------

lolwheel, 2022-06-04:
> [Onewheel_OWIE_BMS_0](https://user-images.githubusercontent.com/39595991/172021873-0ae38706-9ede-4d27-b473-810489a393e1.jpg)

oh one of your cells is wrecked. the voltage is at 2.4v

-------- Original Message --------

an-d-uu, 2022-06-04:
[image]

Here is a shot of the diagnostics page.

an-d-uu, 2022-06-04:
So it has been charging for an hour and the BMS is now reporting 100%, that one cell is still off. Here is a current screen shot. Going to take it for a ride and see how long before it runs out.

[image]

lolwheel, 2022-06-04:
> [Screen Shot 2022-06-04 at 3 10 31 PM](https://user-images.githubusercontent.com/39595991/172024131-b80bffd8-85cb-4ce7-ba8c-f5c926d16191.png)

you need a new battery pack, that's a huge difference in voltages

-------- Original Message --------

an-d-uu, 2022-06-04:
Ok will look at getting a new one, thanks for all your help!

an-d-uu, 2022-06-04:
So after charging the BMS turned the board off at 67% again, so I will take the thing apart make sure all of the soldering points are good and look at getting a new main battery.

[image]

As soon as I plugged it in it reports 72%, so I am wondering if the GT40 I purchased as an external battery pack is causing issues. I will go ahead and close this issue.

closed 2022-06-04 by an-d-uu

lolwheel, 2022-06-04:
> [Screen Shot 2022-06-04 at 3 42 48 PM](https://user-images.githubusercontent.com/39595991/172024979-f9311e22-0a86-4552-8b72-6acbd0c5eac8.png)

This isn't the place to debug the issue, pop into Onewheel discord, a lot of handy people hang in the modding thread. If you haven't had an issue with the battery before, you might have shorted the balancing circuit for the cell while tinkering with the board. That big of a difference in cell voltages is pretty wild

-------- Original Message --------

an-d-uu, 2022-06-05:
Will do, this is the first time I have had issues and was actually wondering how I could debug it.