lolwheel/Owie #49: Board went out while driving

HelxkeblacK, 2022-06-20:
I rode normally my onewheel was at 7% bms and 5% overriding it was my first ride with the chip. Suddenly my board switches off and I'm flying across the street. I immediately looked at the voltage it was at 50V then I see that I got the error 16 I was able to turn the board back on but I carried it home. I don't know why or what happened, I'm fine, I've got a few injuries but nothing is broken.

lolwheel, 2022-06-20:
I've had the same thing happen to me. I never managed to figure out the exact reason but I suspected that the pressure of my front foot could have pressed on the esp8266 (I run mine in the controller board) and maybe disconnect the signal wires for long enough to make the controller throw Err16.

After that happened I also physically removed the reset button from my esp8266 to exclude the chance of it getting accidentally pressed inside of the Onewheel. I just added those instructions explicitly to installation guide.

I don't see anything actionable here unfortunately. Closing the issue.

closed 2022-06-20 by lolwheel

HelxkeblacK, 2022-06-20:
Just for info I already had the problem yesterday that my chip was reset while driving luckily I was driving slowly at the moment I then removed the reset button today so it's definitely something else thanks for the message we'll see what I do it's just creepy.

HelxkeblacK, 2022-06-21:
Can it be that the chip just doesn't get the 5v from the bms in the low percentages? I had to reconnect the wifi all the time from 10% because it was so bad and I accelerated briefly during the nosedive and the voltage dropped to 48-47 maybe the bms didn’t have enough power to power the chip for a moment it went out briefly and gave me the error 16.

lolwheel, 2022-06-21:
I'm pretty sure that's impossible. If Owie lost the power, the BMS itself did too in which case the board would just stay off

-------- Original Message --------