lolwheel/Owie #50: Battery percentage display for cbxr/jwxr

TheoLong, 2022-07-13:
Hi, I'm running JWXR and the cell voltage range from 3v to 4.2v, according to the jwxr battery voltage and percentage chart https://cdn.shopify.com/s/files/1/0468/7252/7001/files/Battery_Voltage.pdf?v=1599258331

The correct percentage lookup table for jwxr/cbxr should be this:
static constexpr uint16_t lookupTableRangeMinMv = 3000;
  static constexpr uint16_t lookupTableRangeMaxMv = 4170;
  static uint8_t lookupTable[11] = {0, 10, 20, 30, 40, 50, 60, 70, 80, 90, 100};

But I understand that the current lookup table maybe is for the stock battery, and stock battery have a different chemistry, which has differnt min/max voltage. So instead of updating the lookup table to this, can we have an option in the UI to set custom min/max cell/total voltage for interpreting percentage? This can cover different types of battery.

lolwheel, 2022-07-13:
Li Ions all have the same voltage characteristics. Having ui to set those options sounds good tho, I'll put it on my back burner.

-------- Original Message --------

TheoLong, 2022-07-13:
According to https://fallman.tech/onewheel-xr/, the XR had NMC, which is not li-ion. Stock BMS (mine is 4209) reported almost 0% at 55v. However, 55v is only at 50% for Jexr/cbxr(li-ion). Having an option to set the percentages lookup allows correct cell voltage display.

Is there a reason why the current lookup table had three 0s at the first three reference point?

lolwheel, 2022-07-13:
NMC is LiIon.

-------- Original Message --------

TheoLong, 2022-07-13:
```
static constexpr uint16_t lookupTableRangeMinMv = 2762;
  static constexpr uint16_t lookupTableRangeMaxMv = 4172;
  static uint8_t lookupTable[11] = {0, 0, 0, 2, 4, 11, 26, 44, 61, 78, 100};
```

Is there a reason why the current lookup table had three 0s at the first three reference point? this effectively drops the percentage to 0 faster.

Li-ion can be drained to 3v safely so I’ll consider 3v as 0%. The current lookup table is not doing that.

lolwheel, 2022-07-13:
I'm not following what or why are we arguing about here.
3v is effectively considered 0, just read the code please.

closing the issue.

-------- Original Message --------

closed 2022-07-13 by lolwheel