lolwheel/Owie #51: Quart battery %

SkaFazz, 2022-07-14:
A fully charged Pint with Chi Quart is reading 94%.

JordanVlieg, 2022-07-27:
Most likely an issue with the battery.  Can you post a screenshot of the OWIE status screen with the individual cell voltages as well as pack voltage visible?

SkaFazz, 2022-07-27:
[image]

This is what I’m getting after a full charge. Thanks 😊

SkaFazz, 2022-07-28:
[image]

this is after a 2 mile ride from full.

lolwheel, 2022-07-29:
Yeah need to rewrite battery cell estimation but don't have time.

JordanVlieg, 2022-08-03:
Unless there is something I am not aware of on the pint where it intentionally doesnt charge to full.  4.14V is not a full battery.  4.2V is full. That battery is indeed 93% full.

JordanVlieg, 2022-08-09:
This is addressed in https://github.com/lolwheel/Owie/pull/55

random1planet, 2023-02-07:
Hi, is this issue resolved as I have a owie chip and 63v quart like pint battery, the BMS reported SOC is 100% so wont charge any further but the overridden SOC value is only 90% charged?

closed 2024-01-03 by lolwheel