lolwheel/Owie #57: Onewheel battery operating "out of safety range"

MikeNilsson, 2022-08-15:
A major update on the OW (XR) was done recently and 150 miles went by with no issues. Now I am getting "out of safety range" message followed by OW will not operate afterwards. It has to be a safety feature but I really do not know how to fix it. This only happens when the weather is over 80 degrees.

When riding I will get below message and after stopping I cannot keep going since the OW will not engage. Shutdown and restart does not matter.

FMM chip removed and replaced with Owie. FMM made my OW nosediving on my with "error 22" messages. 

OWIE installed - working flawless until this latest strange reading.
JWXR installed - working

Original Hardware: 4212
New BMS:                    4209 (original 4212 got fried)
Firmwate:                     Gemini -4155

Battery indicates 32 Farenheit......clearly not reading &/or not working correctly.

I use a:
Iphone 12 pro
IOS 15.6

Any input would be mighty appreciated.
[image]

[image]

lolwheel, 2022-08-16:
Likely not an Owie issue. Probably your BMS or thermistor in the battery pack shat the bed.

This won't help anything but the latest Owie release added battery temp monitoring, you end up with 5 values instead of one. Might be interesting to see what's in there.

Take a screenshot of the "Monitor BMS Data" screen whenever this happens and attach to this issue.

Technically you could override that error on Owie side but I'm hesitant to implement it because it's a liability. You'll lose legitimate over/undertemp warnings.

MikeNilsson, 2022-08-16:
I clearly have an issue since one is reading 0.

So the new 5 temp values are...?!?!?!?! Battery / BMS values. What is what
[image]
 
Here is the BMS Data:
[image]

lolwheel, 2022-08-16:
OK you def have hardware issues.

1. BMS is reporting over/undertemp - the status message has bit number 1 set on it.
2. One of the temp sensors seems to be busted/disconnected as it's reporting 0. I don't know which one is which.
3. Your cells seem to be super out of balance.

I'd suggest opening up the battery enclosure and wiggling the balance lead.

This ain't Owie issue, closing.

closed 2022-08-16 by lolwheel