lolwheel/Owie #59: Hitting the back button in wifi settings saves

yourjelly, 2022-09-03:
Going into the wifi settings and then hitting back saves 'changes' and reboots setting the password to whatever 3 is the default in the password field. This does not appear to be an intentional userflow.

lolwheel, 2022-09-03:
Thanks, I'll look into it in a bit

-------- Original Message --------

closed 2022-12-22 by lolwheel

OnewheelNZ, 2022-12-23:
Hi,

I recently installed the owie with a new CBXR.
Just wondering should the OW app read the correct percentage while riding,
or does it correct itself when stationary? My app shows every time I stop
the board then is recharging.
I did use thinner wire than the 26 gauge recommended, would that mess with
the OWIE reporting the correct percentage to the MB?

Thanks for your help,
Shane

mononen, 2022-12-23:
@OnewheelNZ Firstly, please create a separate issue when reporting problems instead of hijacking an already closed issue.
Second, that is normal as when batteries are no longer under load, their voltage is able to normalize (and thus increase over their voltage under load). Based on the battery calculations being done by voltage, when the pack is under load it will report that it's less full than it actually is. By stopping and letting the cells 'rest' you're giving the pack time to get a reasonable reading.
No your <26AWG wire is not the issue and the board is not recharging magically while riding, that is expected behavior.