lolwheel/Owie #64: XR isn't charging fully after OWIE install

briderx, 2022-10-13:
When I charge my XR (stock battery) after riding, it doesn't seem to charge fully. I have to unplug it and plug it back in for it to charge more. The below pictures show it "fully charged". The first two pics were taken within a minute or two max of each other.
The second two are a few hours charging, after unplugging and plugging it back in. This happens every time. I even swapped chips, thinking there was something wrong with the first one.

Initial Full Charge:
[image]

Unplugged Initial Full Charge:
[image]

2nd Full Charge:
[image]

Unplugged 2nd Full Charge:
[image]

lolwheel, 2022-10-13:
Owie has no say in charging, it cannot control anything.

Owie reports the battery percentage based on voltages and in your case the voltages do indeed fall pretty significantly. The BMS ignores the voltages so that's why you don't see the percentage drop without Owie.

You'll get the same mileage out of the battery with or without Owie. The only thing amiss will be percentage reported.

Are you using a quick charger by any chance? Cells dropping from 4.2 to 4.11 is pretty dramatic, I haven't seen anything close to that myself. Higher current charger might explain this.

anyways nothing Owie can fix except implementing Ah-based percentage counting which I don't have much interest in at this point, closing the issue.

closed 2022-10-13 by lolwheel

briderx, 2022-10-16:
Right on. Np. To answer your question: I don't use the quick charger. Just the regular one. ❤️❤️❤️