lolwheel/Owie #65: XR chip install Thanks I got it figured

OrtonJessop, 2022-10-18:
Xr chip install chart I made I had a hard time understanding the provided one
[image]

lolwheel, 2022-12-22:
Nice thanks, I'll try to put it on the readme page.

bencrundwell, 2023-03-19:
Bump! This would have been really useful to find last week! Can you add it to the readme please

bfaught, 2023-05-01:
This was very helpful, thank you for posting it! Few photos of how I fit it in an XR with a CBXR by CHI Battery Systems. Has worked flawlessly on a 4211 board for the couple rides I have been on since. I did need to extend the white wire a few inches but other than that it fit perfectly. 

[image]

[image]

[image]

OrtonJessop, 2023-05-01:
Orton Jessop