lolwheel/Owie #66: % reading setting off bms error

Kikiobi, 2022-12-15:
Why is 4.14 considered 100%? I’m having an issue when battery reaches 4.2, all the cells start burning off until it gets down to 4.1

Kikiobi, 2022-12-22:
It also shut the board off completely @ 7%, causing a nose dive.

lolwheel, 2022-12-22:
Owie only modifies the reported battery percentage. It physically has no control over what your BMS or controller do.

4.14 is reported as 100% for precisely the reason that after fully charging batteries and leaving them alone, the voltage will drop some.
Board shuts off whenever BMS decides to kill the power. Once again Owie has no say in when this happens. The 7% it read is based on voltage it sees. Chances are your battery pack is old, or maybe you were riding in cold so the voltage dropped below the BMS failsafe level. You should have gotten captain morgan before the board shutoff though. You sure you didn't try riding it after that?

Anyways, there's nothing actionable for me here, closing the issue.

closed 2022-12-22 by lolwheel