lolwheel/Owie #70: Green wire

OrtonJessop, 2023-06-11:
What would happen if you connected the green wire that gets left loose in the same spot as the other end of the green wire on the chip?

Kristowsa, 2023-06-12:
Ha I've thought the same thing myself using a diode inline

Mr-HaleYa, 2023-09-23:
The green wire is the other half of the communication from the BMS to the Board. By cutting and leaving disconnected the Owie can pretend to be the BMS. The Owie can read the data from the BMS off the white wire, spoof what it needs then modulate the white and green out to the Board. Reconnecting it is not needed. Adding a diode inline just blocks the flow of electricity, rendering the line useless, the same as not connecting it. Connecting it to the Owie in the same spot the line to the Board is connected would cause both the BMS and Owie to try sending data on the line to the Board. most likely it would all come through as gibberish and an error 16.
Just leave it disconnected and wrapped in tape or heat shrink.

lolwheel, 2024-01-03:
Mr HaleYa is spot on.

closed 2024-01-03 by lolwheel