lolwheel/Owie #78: Incorrect Battery % vs Battery Voltage

qwerty84532, 2024-02-04:
Apologies if this is in the wrong place (please let me know where to post if needed).

Onewheel XR (4212/4161; cbxr; OWIE).

OWIE was working fine with some lag in showing correct battery % in onewheel app. I noticed that OWIE showed version 0.0.1. I decided to update to 1.4.3.

Now the OWIE status reports significantly incorrect values (see attached screengrab) - Battery Voltage 57.30V (which is 70+%), but BMS reported SOC - 43% and Overide SOC 21%.

Did I update with the wrong firmware?  Any guidance would be welcome.  Thank you

[image]
[image]

bencrundwell, 2024-02-07:
It looks to me like you might have a bad cell sat at 3.71v. Try a balance charge by leaving it on charge overnight and see if the cells are more similar and the % more accurate? Its possible the BMS uses the lowest cell voltage for safety

qwerty84532, 2024-02-07:
Ben,
Yes, that cell is playing up. I did do the overnight balance and all the cells were pretty balanced at 4.2V if I remember right. But during charging, that cell would always read a little higher than the others until it eventually settled. The battery percentage in the OW app would read closely reflect what the OWIE said the battery voltage was.  Once you start riding there would be a difference, the App showing lower battery percentage.  If you stopped and waited a few minutes, the battery percentage in the OW app would go back up to reflect the actual battery Voltage based  percentage. Once I updated to to 1.4.3, the first thing I noticed was that charging the battery would 'get stuck' around 70%.  I thought that because the later updates changed the voltage lookup tables and would show fully charged at 4.14V (I think) and the potentially bad cell with the higher voltage during charging would trigger the charging to stop. 

When I noticed the discrepancy in OWIE battery voltage vs the shown charge% in OWIE and app, I downgraded to 1.4.2 and 1.4.0 with the same results. Once I downgraded to 1.0.0, the OWIE and OW app reported battery % reflected the correct percentage based on the battery voltage.

Thanks for your time and feedback.
Mick

bencrundwell, 2024-02-09:
Hmmm, I'm not sure what more I can offer I'm afraid. I'm still relatively new to the project myself and am in a similar place to where you were initially. The App % is going down too fast and then recovering when I stop. My logs have lots of charging stops in them!

Today I've pulled the latest code as it looks like the main developer was working on a coulomb counter approach and more dynamic battery capacity detection, but haven't been out for a ride to test it yet.  Annoyingly I'm still on the stock battery and use OWIE for other reasons, so I actually do just want the battery to be passed across to the controller and not modified really

FAB1150, 2024-11-19:
I'm encountering the same thing, and it's a bit annoying especially on an older battery as it significantly reduces the already not great range.
All cells are fine in my battery.

[image]