lolwheel/Owie #80: Hardware versions beyond 5314 and Software 5050

bwees, 2024-03-21:
Hello, I am purchasing a Pint that has a hardware revision beyond 5314 and has been upgraded to the latest firmware already. Does Owie work with the newer hardware versions and the latest update as of now? Will there be any issues installing the new battery (such as the firmware bricking when unplugging BMS like on the GT)? 

Thanks

mretich, 2024-04-21:
Hi,
there is basically NO guarantee that it will work, sorry.
What I can tell you ist, that I have a XR with the OWIE FW updated to the newest FM firmware (the haptic feedback fw) and have NO issue at all.
But I have an older HW Rev. as you, so the only way is to try it out...

Relys, 2024-05-02:
I've pretty much just ditched FM hardware and converted my boards to VESC. Stock controllers are dangerous IMO anyways, since the mosfets are to underspeced and then on top of it all with the haptic buzz update you loose a bit of topspeed cuz that action cuts into the already limited max current. 

If you want to hang onto your BMS with a discharge bypass I wrote the BMS parsing code in LispBM which runs directly on a VESC Express and allows you to connect and read cell voltage data and temps directly in the VESC App. :)

https://github.com/Relys/vesc_pkg/blob/float-accessories/float_accessories/float-accessories.lisp#L692